evcc-io/optimizer #15: Service crash

andig, 2025-09-05:
Seeing this crash in evopt.evcc.io:

```
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask/app.py", line 1498, in __call__
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    return self.wsgi_app(environ, start_response)
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]           ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask/app.py", line 1473, in wsgi_app
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    response = self.full_dispatch_request()
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask/app.py", line 880, in full_dispatch_request
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    rv = self.dispatch_request()
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask/app.py", line 865, in dispatch_request
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask_restx/api.py", line 402, in wrapper
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    resp = resource(*args, **kwargs)
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask/views.py", line 110, in view
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    return current_app.ensure_sync(self.dispatch_request)(**kwargs)  # type: ignore[no-any-return]
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask_restx/resource.py", line 41, in dispatch_request
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    resp = meth(*args, **kwargs)
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask_restx/marshalling.py", line 244, in wrapper
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    resp = f(*args, **kwargs)
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/app/app.py", line 436, in post
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    result = optimizer.solve()
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/app/app.py", line 315, in solve
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    self.problem.solve(solver)
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    ~~~~~~~~~~~~~~~~~~^^^^^^^^
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/pulp/pulp.py", line 1883, in solve
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    status = solver.actualSolve(self, **kwargs)
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/pulp/apis/coin_api.py", line 112, in actualSolve
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    return self.solve_CBC(lp, **kwargs)
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]           ~~~~~~~~~~~~~~^^^^^^^^^^^^^^
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/pulp/apis/coin_api.py", line 178, in solve_CBC
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    if cbc.wait() != 0:
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]       ~~~~~~~~^^
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/subprocess.py", line 1280, in wait
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    return self._wait(timeout=timeout)
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]           ~~~~~~~~~~^^^^^^^^^^^^^^^^^
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/subprocess.py", line 2066, in _wait
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    (pid, sts) = self._try_wait(0)
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]                 ~~~~~~~~~~~~~~^^^
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/subprocess.py", line 2024, in _try_wait
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    (pid, sts) = os.waitpid(self.pid, wait_flags)
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]                 ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/gunicorn/workers/base.py", line 203, in handle_abort
2025-09-05T14:05:17Z app[908057d9a02328] fra [info]    sys.exit(1)
```

any help appreciated!

andig, 2025-09-05:
> This is the nuclear option.

🤓

ekkea, 2025-09-06:
I have seen cases with the current model size where the branch-and-bound took as long as 30s. These were rather pathologic cases but it can happen. 
Typical cases are where power export remuneration is higher than the power import cost and the optimizer tries all poosible combinations of cheap grid import in some intervals and profitable export in others. 
If it turns out that the milp solver actually times out, please share the input data for analysis.