evcc-io/optimizer #18: Gunicorn doesn't start

andig, 2025-09-06:
After https://github.com/andig/evopt/pull/13, gunicorn no longer starts:

```
[2025-09-06 13:50:21 +0000] [1] [INFO] Starting gunicorn 22.0.0
[2025-09-06 13:50:21 +0000] [1] [INFO] Listening at: http://0.0.0.0:7050 (1)
[2025-09-06 13:50:21 +0000] [1] [INFO] Using worker: sync
[2025-09-06 13:50:21 +0000] [7] [INFO] Booting worker with pid: 7
[2025-09-06 13:50:21 +0000] [8] [INFO] Booting worker with pid: 8
[2025-09-06 13:50:21 +0000] [7] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/gunicorn/arbiter.py", line 609, in spawn_worker
    worker.init_process()
    ~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/workers/base.py", line 134, in init_process
    self.load_wsgi()
    ~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/workers/base.py", line 146, in load_wsgi
    self.wsgi = self.app.wsgi()
                ~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
                    ~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/app/wsgiapp.py", line 58, in load
    return self.load_wsgiapp()
           ~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/app/wsgiapp.py", line 48, in load_wsgiapp
    return util.import_app(self.app_uri)
           ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/util.py", line 371, in import_app
    mod = importlib.import_module(module)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/app/app.py", line 5, in <module>
    from optimizer import Optimizer, OptimizationStrategy, BatteryConfig, TimeSeriesData
ModuleNotFoundError: No module named 'optimizer'
[2025-09-06 13:50:21 +0000] [7] [INFO] Worker exiting (pid: 7)
[2025-09-06 13:50:22 +0000] [1] [ERROR] Worker (pid:7) exited with code 3
[2025-09-06 13:50:22 +0000] [8] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/gunicorn/arbiter.py", line 609, in spawn_worker
    worker.init_process()
    ~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/workers/base.py", line 134, in init_process
    self.load_wsgi()
    ~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/workers/base.py", line 146, in load_wsgi
    self.wsgi = self.app.wsgi()
                ~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
                    ~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/app/wsgiapp.py", line 58, in load
    return self.load_wsgiapp()
           ~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/app/wsgiapp.py", line 48, in load_wsgiapp
    return util.import_app(self.app_uri)
           ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/util.py", line 371, in import_app
    mod = importlib.import_module(module)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/app/app.py", line 5, in <module>
    from optimizer import Optimizer, OptimizationStrategy, BatteryConfig, TimeSeriesData
ModuleNotFoundError: No module named 'optimizer'
[2025-09-06 13:50:22 +0000] [8] [INFO] Worker exiting (pid: 8)
[2025-09-06 13:50:22 +0000] [1] [ERROR] Worker (pid:8) exited with code 3
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/gunicorn/arbiter.py", line 202, in run
    self.manage_workers()
    ~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/arbiter.py", line 571, in manage_workers
    self.spawn_workers()
    ~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/arbiter.py", line 643, in spawn_workers
    time.sleep(0.1 * random.random())
    ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/arbiter.py", line 242, in handle_chld
    self.reap_workers()
    ~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/arbiter.py", line 530, in reap_workers
    raise HaltServer(reason, self.WORKER_BOOT_ERROR)
gunicorn.errors.HaltServer: <HaltServer 'Worker failed to boot.' 3>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/gunicorn", line 7, in <module>
    sys.exit(run())
             ~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/app/wsgiapp.py", line 67, in run
    WSGIApplication("%(prog)s [OPTIONS] [APP_MODULE]", prog=prog).run()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/app/base.py", line 236, in run
    super().run()
    ~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/app/base.py", line 72, in run
    Arbiter(self).run()
    ~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/arbiter.py", line 229, in run
    self.halt(reason=inst.reason, exit_status=inst.exit_status)
    ~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/arbiter.py", line 342, in halt
    self.stop()
    ~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/arbiter.py", line 396, in stop
    time.sleep(0.1)
    ~~~~~~~~~~^^^^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/arbiter.py", line 242, in handle_chld
    self.reap_workers()
    ~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/gunicorn/arbiter.py", line 530, in reap_workers
    raise HaltServer(reason, self.WORKER_BOOT_ERROR)
gunicorn.errors.HaltServer: <HaltServer 'Worker failed to boot.' 3>
```

ekkea, 2025-09-06:
Vermutlich wird optimizer.py nicht gefunden, weil gunicorn in einem anderen Verzeichnis unterwegs ist? kannst du `--chdir /path/to/directory` mitgeben?

lfloeer, 2025-09-06:
Wenn das in Docker passier könnte es daran liegen, dass du optimizer.py nicht in das Image kopierst. Im Dockerfile wird nur app.py kopiert.