evcc-io/optimizer #21: Recover from crashed worker

andig, 2025-09-21:
```
2025-09-21T08:13:38Z app[908057d9a02328] fra [info][2025-09-21 08:13:38 +0000] [649] [CRITICAL] WORKER TIMEOUT (pid:7180)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info][2025-09-21 08:13:38 +0000] [7180] [ERROR] Error handling request /optimize/charge-schedule
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]Traceback (most recent call last):
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/gunicorn/workers/sync.py", line 135, in handle
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    self.handle_request(listener, req, client, addr)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/gunicorn/workers/sync.py", line 178, in handle_request
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    respiter = self.wsgi(environ, resp.start_response)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask/app.py", line 1498, in __call__
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    return self.wsgi_app(environ, start_response)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]           ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask/app.py", line 1473, in wsgi_app
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    response = self.full_dispatch_request()
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask/app.py", line 880, in full_dispatch_request
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    rv = self.dispatch_request()
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask/app.py", line 865, in dispatch_request
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask_restx/api.py", line 402, in wrapper
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    resp = resource(*args, **kwargs)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask/views.py", line 110, in view
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    return current_app.ensure_sync(self.dispatch_request)(**kwargs)  # type: ignore[no-any-return]
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask_restx/resource.py", line 41, in dispatch_request
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    resp = meth(*args, **kwargs)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/flask_restx/marshalling.py", line 244, in wrapper
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    resp = f(*args, **kwargs)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/app/app.py", line 171, in post
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    result = optimizer.solve()
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/app/optimizer.py", line 276, in solve
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    self.problem.solve(solver)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    ~~~~~~~~~~~~~~~~~~^^^^^^^^
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/pulp/pulp.py", line 2092, in solve
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    status = solver.actualSolve(self, **kwargs)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/pulp/apis/coin_api.py", line 140, in actualSolve
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    return self.solve_CBC(lp, **kwargs)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]           ~~~~~~~~~~~~~~^^^^^^^^^^^^^^
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/pulp/apis/coin_api.py", line 218, in solve_CBC
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    if cbc.wait() != 0:
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]       ~~~~~~~~^^
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/subprocess.py", line 1280, in wait
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    return self._wait(timeout=timeout)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]           ~~~~~~~~~~^^^^^^^^^^^^^^^^^
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/subprocess.py", line 2066, in _wait
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    (pid, sts) = self._try_wait(0)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]                 ~~~~~~~~~~~~~~^^^
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/subprocess.py", line 2024, in _try_wait
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    (pid, sts) = os.waitpid(self.pid, wait_flags)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]                 ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]  File "/usr/local/lib/python3.13/site-packages/gunicorn/workers/base.py", line 203, in handle_abort
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    sys.exit(1)
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]    ~~~~~~~~^^^
2025-09-21T08:13:38Z app[908057d9a02328] fra [info]SystemExit: 1
2025-09-21T08:13:38Z app[908057d9a02328] fra [info][2025-09-21 08:13:38 +0000] [7180] [INFO] Worker exiting (pid: 7180)
```

leads to:

```
2025-09-21T08:19:25Z app[908057d9a02328] fra [info][2025-09-21 08:19:25 +0000] [649] [ERROR] Worker (pid:7339) was sent SIGKILL! Perhaps out of memory?
2025-09-21T08:19:25Z app[908057d9a02328] fra [info][2025-09-21 08:19:25 +0000] [7364] [INFO] Booting worker with pid: 7364
2025-09-21T08:19:26Z proxy[908057d9a02328] fra [error][PR03] could not find a good candidate within 20 attempts at load balancing. last error: [PR01] no known healthy instances found for route tcp/443. (hint: is your app shut down? is there an ongoing deployment with a volume or are you using the 'immediate' strategy? have your app's instances all reached their hard limit?)
2025-09-21T08:19:28Z proxy[908057d9a02328] fra [error][PR03] could not find a good candidate within 1 attempts at load balancing. last error: [PC05] timed out while connecting to your instance. this indicates a problem with your app (hint: look at your logs and metrics)
2025-09-21T08:19:29Z proxy[908057d9a02328] fra [error][PR04] could not find a good candidate within 20 attempts at load balancing
2025-09-21T08:19:32Z proxy[908057d9a02328] arn [error][PR04] could not find a good candidate within 20 attempts at load balancing
2025-09-21T08:20:02Z app[908057d9a02328] fra [info][2025-09-21 08:20:02 +0000] [649] [CRITICAL] WORKER TIMEOUT (pid:7336)
2025-09-21T08:20:04Z app[908057d9a02328] fra [info][2025-09-21 08:20:04 +0000] [649] [ERROR] Worker (pid:7336) was sent SIGKILL! Perhaps out of memory?
2025-09-21T08:20:04Z app[908057d9a02328] fra [info][2025-09-21 08:20:04 +0000] [7369] [INFO] Booting worker with pid: 7369
```

Something seems to eat all resources until the machine is exhausted.

lfloeer, 2025-09-21:
Ich hatte mehr an Statistiken auf Host-Ebene gedacht. Wenn du Prometheus verwendest, sollte man da ja auch Metriken der VM bekommen, auf der Docker läuft.

lfloeer, 2025-09-21:
Hostest du das auf fly.io? Ich hab da keine Erfahrung mit, hab aber das hier gefunden, was ganz sinnvoll klingt 

https://fly.io/docs/monitoring/metrics/#instance-memory-fly_instance_memory_

andig, 2025-09-21:
```
curl https://api.fly.io/prometheus/evcc/ \
--header "Authorization: FlyV1 ..." \
--header 'Authorization: Bearer ...'
404 page not found
```

lfloeer, 2025-09-21:
Dann ist’s wohl eher kein schleichender Memory Leak. Man kann in den Logs glaube ich auch erkennen, dass in der Tat sofort neue worker gestartet werden, dann aber wieder ein worker gekillt wird (die PIDs sind immer andere). 

Wie sieht es denn aus mit den Anzahl an Verbindungen und aktiven requests? Nicht, dass das nachher an einem Request liegt, der jede Instanz direkt crashed (warum auch immer) und der Client den immer wieder schickt und so nach und nach alle Instanzen abwechselnd killt… ist aber nur ne Vermutung.

lfloeer, 2025-10-07:
Ich hatte die Tage immer mal wieder 502/503er auf dem Service. Sind das wieder die oben beschriebenen Fehler oder ist es was neues?

andig, 2025-10-07:
I kanns nicht mit Sicherheit sagen. Maschine ist jetzt (15m Slots gibts mehrere) auf 4 CPU 1 GB upgraded.

andig, 2025-10-08:
Just noticed machine at 100% CPU over all cores and load balancing failing. htop shows a number of running CBC processes. I couldn't figure out immediately if there were any old processes or all recently started. Wondering if we can limit CBC runtime somehow?

lfloeer, 2025-10-16:
Hast du mal das in #34 eingeführte time limit probiert?

lfloeer, 2025-10-16:
Ne, aber wenn der optimizer nach 30 Sekunden keine Lösung findet, greift das timeout des Webservers und die Verbindung wird abgebrochen ohne dass der optimizer gestoppt wird. Deswegen setzen wir das optimizer timeout auf 30 Sekunden damit hoffentlich nach dem Verbdindungsabbruch auch der optimizer beendet wird. Das mit den Threads sollte für deinen Fall nicht so wichtig sein.

lfloeer, 2025-10-16:
Würde ich erwarten. Sind viermal so viele Variablen.

CiNcH83, 2025-10-17:
Seit der Umstellung scheint erstmal Ruhe zu sein. Ich beobachte weiter...

Macht es evtl. Sinn, die CBC Laufzeit etwas zu profilen? Evtl. erstmal mit einem einfachen `min`, `max`, `avg`.

CiNcH83, 2025-10-17:
Der Fehler war jetzt wieder da:
```
[site ] ERROR 2025/10/17 08:10:00 optimizer: Post "http://localhost:7050/optimize/charge-schedule": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] ERROR 2025/10/17 08:10:15 optimizer: Post "http://localhost:7050/optimize/charge-schedule": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] ERROR 2025/10/17 08:10:30 optimizer: Post "http://localhost:7050/optimize/charge-schedule": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
```

Mal schauen, ob sich die `cbc` Prozesse mit der Zeit wieder aufschaukeln...

CiNcH83, 2025-10-19:
I have had no more problems since Friday noon with the above settings, neither HTTP nor worker (`Not Solved`) timeouts. I think what did the trick in my case was the `threads=2`, most likely cutting execution time in half.

andig, 2025-10-23:
Mhhm. It doesn‘t look all bad then. Not sure why I‘m getting timeouts.

CiNcH83, 2025-10-25:
Jetzt hatte ich gerade konzentriert ein paar HTTP Timeouts:
```
[site ] ERROR 2025/10/25 11:17:03 optimizer: Post "http://localhost:7050/optimize/charge-schedule": context deadline exceeded
[site ] ERROR 2025/10/25 11:20:03 optimizer: Post "http://localhost:7050/optimize/charge-schedule": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] ERROR 2025/10/25 11:25:18 optimizer: Post "http://localhost:7050/optimize/charge-schedule": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[site ] ERROR 2025/10/25 11:28:18 optimizer: Post "http://localhost:7050/optimize/charge-schedule": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] ERROR 2025/10/25 11:31:18 optimizer: Post "http://localhost:7050/optimize/charge-schedule": context deadline exceeded
[site ] ERROR 2025/10/25 11:34:18 optimizer: Post "http://localhost:7050/optimize/charge-schedule": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] ERROR 2025/10/25 11:37:18 optimizer: Post "http://localhost:7050/optimize/charge-schedule": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] ERROR 2025/10/25 11:40:18 optimizer: Post "http://localhost:7050/optimize/charge-schedule": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[site ] ERROR 2025/10/25 11:43:18 optimizer: Post "http://localhost:7050/optimize/charge-schedule": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] ERROR 2025/10/25 11:46:18 optimizer: Post "http://localhost:7050/optimize/charge-schedule": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] ERROR 2025/10/25 11:49:18 optimizer: Post "http://localhost:7050/optimize/charge-schedule": context deadline exceeded
[site ] ERROR 2025/10/25 11:52:18 optimizer: Post "http://localhost:7050/optimize/charge-schedule": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] ERROR 2025/10/25 11:55:18 optimizer: Post "http://localhost:7050/optimize/charge-schedule": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
```

evopt Log zeigt keine Unzulänglichkeiten:
```
Pi@raspberrypi5:~ $ sudo docker logs evopt
[2025-10-22 17:51:30 +0200] [1] [INFO] Starting gunicorn 23.0.0
[2025-10-22 17:51:30 +0200] [1] [INFO] Listening at: http://0.0.0.0:7050 (1)
[2025-10-22 17:51:30 +0200] [1] [INFO] Using worker: sync
[2025-10-22 17:51:30 +0200] [7] [INFO] Booting worker with pid: 7
[2025-10-23 20:28:03 +0200] [7] [INFO] Autorestarting worker after current request.
[2025-10-23 20:28:03 +0200] [7] [INFO] Worker exiting (pid: 7)
[2025-10-23 20:28:03 +0200] [731] [INFO] Booting worker with pid: 731
[2025-10-24 22:31:48 +0200] [731] [INFO] Autorestarting worker after current request.
[2025-10-24 22:31:48 +0200] [731] [INFO] Worker exiting (pid: 731)
[2025-10-24 22:31:48 +0200] [1455] [INFO] Booting worker with pid: 1455
```

evopt Konfig:
```
    environment:
      - TZ=Europe/Berlin
      - OPTIMIZER_NUM_THREADS=2
      - OPTIMIZER_TIME_LIMIT=25
      - GUNICORN_CMD_ARGS=--workers 1 --max-requests 720
```

Angehängt noch das Trace-Log. Denke aber nicht, dass da der Request dran schuld ist.

`htop` und `ps aux` zeigen keine hängen gebliebenen CBC Prozesse.

[evcc-20251025-115754-trace.log](https://github.com/user-attachments/files/23141177/evcc-20251025-115754-trace.log)

andig, 2025-10-30:
Der Restart selbst scheint nicht das Problem zu sein (/tmp leer). Heute allerdings war's wieder soweit: 600MB in /tmp. Wir brauchen also irgendeine Art von Aufräummechanismus.