evcc-io/optimizer #22: Add incentive for continous battery usage

andig, 2025-09-21:
Batteries are turned on and off (see 2nd day 14:00):

[image]

```json
{
  "batteries": [
    {
      "c_max": 11040,
      "c_min": 1380,
      "charge_from_grid": true,
      "d_max": 0,
      "p_a": 0.0002896641,
      "s_initial": 1253,
      "s_max": 17900,
      "s_min": 0
    },
    {
      "c_max": 1200,
      "c_min": 0,
      "d_max": 800,
      "p_a": 0.0002896641,
      "s_initial": 1574.4,
      "s_max": 3763.2,
      "s_min": 384
    }
  ],
  "eta_c": 0.9,
  "eta_d": 0.9,
  "strategy": {
    "charging_strategy": "charge_before_export"
  },
  "time_series": {
    "dt": [328, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600, 3600],
    "ft": [95.18586, 1143.275, 1186.45, 1186.125, 1202.85, 1110.1, 862.75, 468.775, 144.875, 19.175, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 3.2, 75.725, 405.35, 1124.425, 2134.725, 2775.475, 2861.375, 2972.525, 2950.85, 2579.8, 1969.025, 1051.4, 283.65, 28.825, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 8.475, 208.125, 0],
    "gt": [52.184082, 520.1184, 510.80713, 506.4717, 462.472, 386.8251, 432.1674, 488.36887, 433.5858, 427.3584, 297.5788, 227.20483, 202.27895, 197.22377, 199.6096, 196.37796, 193.69164, 200.64168, 216.88495, 326.37585, 357.89276, 435.6917, 506.67163, 640.7322, 521.8408, 520.1184, 510.80713, 506.4717, 462.472, 386.8251, 432.1674, 488.36887, 433.5858, 427.3584, 297.5788, 227.20483, 202.27895, 197.22377, 199.6096, 196.37796, 193.69164, 200.64168, 216.88495, 326.37585, 357.89276, 435.6917, 506.67163],
    "p_E": [0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012, 0.00012],
    "p_N": [0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251, 0.0003251]
  }
}
```

ekkea, 2025-10-03:
would it make sense to define a priority sequence for the batteries in terms of which one to charge first if it makes no economic difference?

andig, 2025-10-03:
Yes, absolutely!

lfloeer, 2025-10-07:
I was thinking about this as well. Would it be possible to add a penalty that penalizes switching between off and on, i.e. minimize the number of transitions from 0 power to >0 power?  This could be physically motivated by the cost associated with switching a device on.

ekkea, 2025-10-07:
I do not think that there is a specific cost for switching battery (dis)charging on and off, not in efficiency nor in lifetime. These costs are known from mechanical systems like motors, engines, compressors - in many cases because the lubrication at startup is not ideal. The only problem remaining in power electronics will be thermal cycling but that is not related to on/off switching but load transients in general.
Another issue here is that with 15min time steps we are far away from the actual load fluctuation frequency in electric systems. Transients in electrical installations happen to occur in a millisecond time frame (e.g. switch on a cooking plate where only the grid impedance will limit the current rise). The inverter pwm control cycles are even magnitudes below that at 100µs and much less. Switching every 15 min or not will not make a difference.
Finally, what the evcc system will use for system control decisions is only the result's first time step. The optimization will be recomputed frequently, long before we ever reach into the middle of the optimization's time horizon (this is the principle of model based control). Therefore, it really comes down to two aspects:
1. we can prioritize one battery over another in terms of which one gets charged first. This can actually affect the result for the first time step that is taken for system control decisions.
2. we can help the user understand the optimization result better when charging periods are shown in consistent blocks.
From an implementation point of view the charging priorities are much easier to handle than penalizing switching, likely it is the same for the computational load. I think the results will look pretty similar in most of the use cases.
Very long story short: I would go for priorities rather than penalizing switching.

lfloeer, 2025-10-07:
Thank you for the detailed elaboration. Obviously you have thought about this quite some more than me. Your suggestion sounds good to me.

ekkea, 2025-10-20:
@andig, should we prioritize just by sequence in the battery array, or would you prefer to add an explicit interface property for that?

closed 2026-01-19 by andig